SBFspot/sbfspot-config #1: ERROR  1146 : able 'sbfspot.config' doesn't exist

danny12321, 2018-09-30:
I Get this error message wheen I try to install and I don't know how to solve it. 

│There is a problem: 
│ ERROR 1146 (42S02) at line 1: Table 'sbfspot.config' doesn't exist

sbf-, 2018-10-01:
When using MySQL, you have to create the SBFspot database and SBFspotUser manually.
I published V1.0.8 today. Retry the installation - it will fail again, but it will leave 2 .sql files in /usr/local/bin/sbfspot.3
I'll update the readme soon.

danny12321, 2018-10-01:
Thank you, I don't have that problem anymore. But now I have the error 'Invalid timezone specified: UTC'. Do you know how I can set a valid timezone?

sbf-, 2018-10-01:
Timezone must be set with `raspi-config` (Localization Options->Change Timezone)

AJErazzor, 2018-10-22:
Getting the same error
 There is a problem:                                              
 ERROR 1146 (42S02) at line 1: Table 'sbfspot.config' doesn't exi
remote mySQL server - user and database created with script, however I added full permissions to user to see if this would help but no

sbf-, 2018-10-22:
Running MySQL server on Linux?
If yes then it's probably a case sensitivity issue (SELECT * FROM SBFspot.Config)

AJErazzor, 2018-10-22:
Yes Linux - will try changing table name and report back

Thanks
Andrew

AJErazzor, 2018-10-23:
I ran a local copy and changed (SELECT * FROM SBFspot.config) to (SELECT * FROM Config)

Get past that but now getting:
/usr/local/bin/sbfspot.3/SBFspot: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version `CXXABI_1.3.9' not found (required by /usr/local/bin/sbfspot.3/SBFspot)
/usr/local/bin/sbfspot.3/SBFspot: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by /usr/local/bin/sbfspot.3/SBFspot)

sbf-, 2018-10-23:
That's a bigger issue ;-)
What image is installed on your RPi?

AJErazzor, 2018-10-23:
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian

sbf-, 2018-10-23:
Good old Jessie ;-)
Only tested on Stretch.

AJErazzor, 2018-10-25:
nope that didnt work - i have upgraded to Stretch and the changed case in the batch file works fine now - just have to work out the SetPlantTime returned an error: -1 issue now but nothing to do with this issue

closed 2018-11-07 by sbf-