SBFspot/sbfspot-config #2: Support for changing Timezone

mrBussy, 2019-01-03:
Hi,

I have the timezone of my RPi set on Etc/UTC. This is not supported by the configuration tool. When I want to change this to my actual timezone (Europe/Amsterdam) this is not possible. Since the Etc/UTC is not recognized my setup will not finish.

There is a workaround available by creating the configuration file, change this by hand and reload the setup with the configuration file. 

Please add the option to change the timezone configuration within the Menu.

sbf-, 2019-01-04:
sbfspot-config reads the timezone from `/etc/timezone` and gets a default location (see [tzlatlon](https://github.com/SBFspot/sbfspot-config/blob/master/tzlatlon))
Changing the timezone is not supported by the tool because raspi-config does this, and I don't want to duplicate this functionality in sbfspot-config
So, you can fill in your location, disregarding the timezone.
Location is used for sunrise/sunset calculation.

mrBussy, 2019-01-07:
Hi, I probably was not clear enough. I'm not talking about configuration of the timezone on Linux itself. I'm talking about the time zone in the configuration of sbfspot-config. 

As you already explain the time zone is read from /etc/timezone. With my setup this is Etc/UTC. When running the sbfspot-config tool this is placed in the configuration file. And This than provides an error for the installation performed by the script. It will search for Etc/UTC in tzlation and there the location does not exist resulting in a stop of the installation file.

So adding Etc/UTC as timezone to the file will solve the issue (or let the user change it in the menu and write that to the sbf config file).

commit f727b5c5 (2024-06-20): V1.3.7 (Fixes #2)
closed 2024-06-20 by sbf-