SBFspot/sbfspot-config #5: installation aborted: libstdc++ using depreciated version

prein2, 2019-03-21:
Installation fails:

/usr/local/bin/sbfspot.3/SBFspot: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: v                                                         ersion `CXXABI_1.3.9' not found (required by /usr/local/bin/sbfspot.3/SBFspot)
/usr/local/bin/sbfspot.3/SBFspot: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: v                                                         ersion `GLIBCXX_3.4.21' not found (required by /usr/local/bin/sbfspot.3/SBFspot)

I am not a programmer, but I found another reference where this issue is solved with the following: put /home/user/anaconda2/lib/libstdc++.so.6.0.24 in LD_PRELOAD enviroment variable.

sbf-, 2019-03-22:
What OS are you running? Raspian Stretch?

prein2, 2019-03-25:
Raspbian Jessie. I Installed using the manual method, so can't test anymore :(

sbf-, 2019-03-26:
No problem....
Precompiled binaries are not compatible with Jessie. See [this issue](https://github.com/SBFspot/SBFspot/issues/260)

closed 2019-03-26 by sbf-