SBFspot/sbfspot-config #6: Upload daemon not updated

mk-au, 2019-05-21:
I had SBFspot 3.4.0 installed.

When I ran the tool, it doesn't appear to have updated the upload daemon based on file times. Should it have done so? what's the best way to update it?

```
/usr/local/bin/sbfspot.3 $ ls -lat
total 1716
-rw-r--r-- 1 root staff    265 May 21 17:51 SBFspotUpload.service
-rw-r--r-- 1 root staff   1532 May 21 17:51 SBFspotUpload.cfg
-rwxr-xr-x 1 root staff    190 May 21 17:51 monthdata
-rwxr-xr-x 1 root staff    181 May 21 17:51 daydata
-rw-r--r-- 1 root staff   2479 May 21 17:51 SBFspot.cfg
-rw-r--r-- 1 root staff 189975 May 21 17:51 TagListNL-NL.txt
-rw-r--r-- 1 root staff 195895 May 21 17:51 TagListIT-IT.txt
-rw-r--r-- 1 root staff 201193 May 21 17:51 TagListFR-FR.txt
-rw-r--r-- 1 root staff 197709 May 21 17:51 TagListES-ES.txt
-rw-r--r-- 1 root staff 187406 May 21 17:51 TagListEN-US.txt
-rw-r--r-- 1 root staff 195265 May 21 17:51 TagListDE-DE.txt
-rw-r--r-- 1 root staff  35855 May 21 17:51 date_time_zonespec.csv
-rwxr-xr-x 1 root staff 371176 May 21 17:51 SBFspot
drwxr-sr-x 2 root staff   4096 Jun 13  2018 .
-rw-r--r-- 1 root staff   1854 May 15  2018 SBFspotUpload.default.cfg
-rwxr-xr-x 1 root staff 120996 May 15  2018 SBFspotUploadDaemon
-rw-r--r-- 1 root staff   1142 May 15  2018 Update_340_SQLite.sql
-rw-r--r-- 1 root staff   5935 May 15  2018 SBFspot.default.cfg
drwxrwsr-x 3 root staff   4096 May 15  2018 ..
```

mk-au, 2019-05-21:
Actually I think I might have found why this wasn't updated.

It was still running.

Perhaps the script needs to check if its running and stop it before updating.

Just noticed this in the logs:
```
Copying daemon...
cp: cannot create regular file './SBFspotUploadDaemon': Text file busy
```

mk-au, 2019-05-21:
I stopped the SBFspotUpload service then re-ran the script. This time it completed and looks like it's updated properly. No more error of being unable to copy the file.

```
[17:54:41] INFO: Starting SBFspotUploadDaemon Version 1.1.0
[17:54:41] INFO: Starting Daemon...
[18:25:40] INFO: SBFspotUploadDaemon Version 2.0.0
[18:25:40] INFO: Starting Daemon...
```

mk-au, 2019-05-21:
For ref, here's the directory listing after re-running the script with the upload daemon stopped:

```
/usr/local/bin/sbfspot.3 $ ls -lat
total 1712
-rw-r--r-- 1 root staff    265 May 21 18:25 SBFspotUpload.service
-rw-r--r-- 1 root staff   1532 May 21 18:25 SBFspotUpload.cfg
-rwxr-xr-x 1 root staff 116824 May 21 18:25 SBFspotUploadDaemon
-rwxr-xr-x 1 root staff    190 May 21 18:25 monthdata
-rwxr-xr-x 1 root staff    181 May 21 18:25 daydata
-rw-r--r-- 1 root staff   2479 May 21 18:25 SBFspot.cfg
-rw-r--r-- 1 root staff 189975 May 21 18:25 TagListNL-NL.txt
-rw-r--r-- 1 root staff 195895 May 21 18:25 TagListIT-IT.txt
-rw-r--r-- 1 root staff 201193 May 21 18:25 TagListFR-FR.txt
-rw-r--r-- 1 root staff 197709 May 21 18:25 TagListES-ES.txt
-rw-r--r-- 1 root staff 187406 May 21 18:25 TagListEN-US.txt
-rw-r--r-- 1 root staff 195265 May 21 18:25 TagListDE-DE.txt
-rw-r--r-- 1 root staff  35855 May 21 18:25 date_time_zonespec.csv
-rwxr-xr-x 1 root staff 371176 May 21 18:25 SBFspot
drwxr-sr-x 2 root staff   4096 Jun 13  2018 .
-rw-r--r-- 1 root staff   1854 May 15  2018 SBFspotUpload.default.cfg
-rw-r--r-- 1 root staff   1142 May 15  2018 Update_340_SQLite.sql
-rw-r--r-- 1 root staff   5935 May 15  2018 SBFspot.default.cfg
```

sbf-, 2019-05-22:
> Actually I think I might have found why this wasn't updated.
> 
> It was still running.
> 
> Perhaps the script needs to check if its running and stop it before updating.
> 
> Just noticed this in the logs:
> 
> ```
> Copying daemon...
> cp: cannot create regular file './SBFspotUploadDaemon': Text file busy
> ```

Termination of the daemon is handled in the script...
`echo "Stopping daemon..."`

sbf-, 2019-05-22:
Daemon stop code never gets executed due to the dot inside the path
`if [ -f "$dir_program./SBFspotUploadDaemon" ]; then`
Should be
`if [ -f "$dir_program/SBFspotUploadDaemon" ]; then`

I'll fix it

commit 65d51c34 (2019-05-22): Fixed #6
closed 2019-05-22 by sbf-