SBFspot/sbfspot-config #8: Duplicate upload service symlink

mk-au, 2019-05-21:
From v3.4.0 I had a symlink located in the base directory.

sbfspot-config doesnt appear to have checked or removed this, but created a new one:

Here's from the sbfspot-config log, looks like it created a new one:

`Created symlink /etc/systemd/system/multi-user.target.wants/SBFspotUpload.service → /usr/local/bin/sbfspot.3/SBFspotUpload.service.
`

Here's my original:
```
/etc/systemd/system $ ls -la
lrwxrwxrwx  1 root root   46 Jun 13  2018 SBFspotUpload.service -> /usr/local/bin/sbfspot.3/SBFspotUpload.service
```

mk-au, 2019-05-21:
Actually, I may be wrong about this being a duplicate, as when I reran the script to correct my other issue with the upload service, it recreated the first symlink too.

Here's both the update logs for ref:

When I first did it, it logged:
`Created symlink /etc/systemd/system/multi-user.target.wants/SBFspotUpload.service → /usr/local/bin/sbfspot.3/SBFspotUpload.service.`

Thinking the first one was a duplicate, I removed it.

On the second run, it didn't do the above file but it did do:
`Created symlink /etc/systemd/system/SBFspotUpload.service → /usr/local/bin/sbfspot.3/SBFspotUpload.service.`

sbf-, 2019-05-22:
symlink is created by systemctl (not sbfspot-config)

closed 2019-05-22 by sbf-