SBFspot/sbfspot-config #10: Debian Buster for raspi

agricolaXIII, 2019-08-10:
Hello,
last week I had upgradet to Debian Buster.
After the upgrade SBFSpot has damaged the system, so I must power of my raspian. I'd try that three times. Since I had disabled the SBFSpot exe file, the system run stable.
Tomorow I'd try to reinstall SBFSpot, and it's the same problem.
Is there a planning to upgrade SBFSpot for Debian Buster? Or is there a problem on my new Debian and the latest version is running on other Buster-Systems?

Greetings 
Jens

sbf-, 2019-08-10:
For SBFspot there are no known issues.
For the upload daemon to PVO, there is this temporary [workaround](https://github.com/SBFspot/SBFspot/issues/295)

deagel, 2020-02-16:
Hi,
is it possible to add the workaround to sbfspot-config inkl. a mariadb / mysql supported version?

thanks

dennis

jvlerick, 2020-03-25:
After upgrading my system to buster SBFspot no longer reports anything to my Domoticz.
When testing with "/usr/local/bin/sbfspot.3/SBFspot -v -finq -nocsv" I get this error: "error while loading shared libraries: libboost_date_time.so.1.62.0: cannot open shared object file: No such file or directory". Apparently with upgrading to Buster libboost version is now 1.67. Is there any way to solve this?

sbf-, 2020-05-25:
Fixed in V1.1.0

closed 2020-05-25 by sbf-