SBFspot/sbfspot-config #13: Outdated versions of files cause failure to install??

tccack, 2019-10-25:
I have attempted to install SBFSpot via the auto installer and cant get past this error with a couple of files (version issues). I've tried the lastest version and also 3.5.0. Error is the same unfortunately.
Any ideas to upgrade to the required versions?

curl -s https://raw.githubusercontent.com/sbfspot/sbfspot-config/master/sbfspot-config | sudo bash -s 3.5.0
SBFspot Configuration Tool V1.0.15
Running on Raspberry Pi 3 Model B Rev 1.2
Checking whiptail... Installed.
architecture=armhf
Raspbian GNU/Linux 8.0 (jessie)
username=pi
userhome=/home/pi
Minimum SBFspot version supported by this tool is 3.5.0
Request to install version 3.5.0
Searching for existing configuration(s)...
Reading defaults from /var/tmp/sbfspot.3/sbfspot.defaults... Done
read_defaults:/home/pi/smadata
Saving defaults to /var/tmp/sbfspot.3/sbfspot.defaults... Done
Performing some checks before installation... Done
Downloading https://github.com/sbfspot/sbfspot/releases/download/V3.5.0/sbfspot-sqlite-arm-linux.tar.gz
Copying files...
SBFspot.db exists
Checking database...
Schema Version=1
Configuring SBFspot... Done
Running SBFspot...
/usr/local/bin/sbfspot.3/SBFspot: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version `CXXABI_1.3.9' not found (required by /usr/local/bin/sbfspot.3/SBFspot)
/usr/local/bin/sbfspot.3/SBFspot: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by /usr/local/bin/sbfspot.3/SBFspot)

Thanks

sbf-, 2019-10-25:
The precompiled files don't run on jessie.
Either upgrade your pi or follow the wiki to compile sbfspot from.scratch.

sbf-, 2019-10-25:
Keep in mind you're already 2 versions behind.

closed 2019-10-25 by sbf-