SBFspot/sbfspot-config #15: crontab deleted

ThomasH-W, 2020-01-19:
First at all thanks for the tool. Simple to install and easy to use.

Please add a note to your readme or even better a hint durin gthe install routine in regards to deleting cron jobs.
	# delete all jobs
	crontab -r -u "$username" 2>/dev/null

I was installing SBFspot as an additional feature and would have never expect that my existing cron jobs would be erased.

sbf-, 2020-01-19:
Cronjobs not related to sbfspot should be retained.
`# Get current cron jobs without SBFspot related ones
	cron=$(crontab -l -u "$username" 2>/dev/null|grep -Eqv "(^$|SBFspot|daydata|monthdata)")`

ThomasH-W, 2020-01-19:
I had this behaviour on two raspberry pi#s yesterday.
Looking at crontab -e, basically all prior lines were gone incl. the standard comments at the top.

cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

sbf-, 2020-01-19:
I never heard of issues with that part of the script.
I’ll retest it.

sbf-, 2020-01-19:
Well, this is a nasty bug... Don't know why this was never discovered before.
grep arguments shouldn't have -q
```
grep -Eqv "(^$|SBFspot|daydata|monthdata)"
```
Sorry for the inconvenience.

closed 2020-01-19 by sbf-

ThomasH-W, 2020-01-19:
Cool. Thanks for the quick fix.

commit 8c0041a6 (2020-01-19): Fixed #15