SBFspot/sbfspot-config #21: New install with sbfspot-config fails to unavailable version 3.7.0

HannikMathe, 2020-06-21:
After a fresh install of Raspberry Pi OS Lite on my RaspberryPi 3b+ , 
I tried to install SBFspot via the sbfspot-config file. 
When I do so I get this error message:

>   Request to install version 3.7.0
>   Version 3.7.0 is not available for installation.
>   Select one of 3.6.0 3.7.0

So I requested version 3.7.0, but version 3.7.0 is not available, and i should choose version 3.7.0 .
Is there anything I don't get or is this a bug?

What I did:
1. Get my Raspberry Pi 3b+
2. Insert SD Card with fresh Raspberry Pi OS Lite (Released: 2020-05-27) and blank ssh file on it
3. Insert LAN and power cable
4. Open terminal in Linux and connect to Pi via: `ssh pi@192.168.xxx.xx` with standard password
5. Execute `curl -s https://raw.githubusercontent.com/sbfspot/sbfspot-config/master/sbfspot-config | sudo bash -s 3.7.0`
6. Get answer: 
> SBFspot Configuration Tool V1.1.1
> Running on Raspberry Pi 3 Model B Plus Rev 1.3
> Checking whiptail... Installed.
> architecture=armhf
> Raspbian GNU/Linux 10 (buster)
> username=pi
> userhome=/home/pi
> Minimum SBFspot version supported by this tool is 3.5.0
> Request to install version 3.7.0
> Version 3.7.0 is not available for installation.
> Select one of 3.6.0 3.7.0

PS: I tried it even with version 3.6.0

sbf-, 2020-06-21:
Hmmm, have to check this. Looks like a bug indeed...

HarvsG, 2020-06-21:
I got the same bug but not on 8c0041a645e698ce23b4dc9d0614fec8fea662f8

HarvsG, 2020-06-22:
I have created a working fork until this is patched:

https://github.com/HarvsG/sbfspot-config

### Run the tool
To install the latest SBFspot release:    
`curl -s https://raw.githubusercontent.com/HarvsG/sbfspot-config/master/sbfspot-config | sudo bash`

To install a specific SBFspot release (e.g. 3.5.0):    
`curl -s https://raw.githubusercontent.com/HarvsG/sbfspot-config/master/sbfspot-config | sudo bash -s 3.5.0`

sbf-, 2020-06-22:
> I think this is the culprit line, `contains()` seems to be returning 1 innapropriately:
> 
> Hope this is helpful @sbf-

Sure, should be `not contains` -- don't know why it worked before

commit d6332852 (2020-06-22): Fix #21
closed 2020-06-22 by sbf-