SBFspot/sbfspot-config #23: Can't enter multiple SystemID's for PVOutput

alainsch, 2020-08-14:
I have multiple SMA invertors and try to enter the PVOutput SID's and invertors s/n in the dialog:

"Enter PVOutput System ID's (comma separated)

2120096127:67000,2120084956:67001"

when pressing OK, I get the message "invalid"

"Enter PVOutput System ID's (comma separated)

2120096127,2120084956"

... fails on install: "Problem with PVOutput API key or SystemID"

PVO enabled, configuring upload daemon...
Device Serial(s): 2120084956
2120096127
Not replacing daemon (unchanged)
Checking SID:2120096127
Unauthorized 401: Missing, invalid or inactive api key information (X-Pvoutput-Apikey)

the API key is the correct one as it is copied and pasted from the pvoutput.org site

What can I do to solve this ?

sbf-, 2020-08-14:
PVoutput system id’s are 67000,67001
You are entering the serial nrs of your inverters

alainsch, 2020-08-14:
ok, that solved it (partly)...

How can I force to bind the system ID's to the correct serial numbers, as I had to enter it as 67001, 67000, otherwise the invertors are swapped

sbf-, 2020-08-14:
No, but that might be an enhancement.

sbf-, 2022-08-27:
Fixed in V1.3.0 (ec5f277)

closed 2022-08-27 by sbf-