SBFspot/sbfspot-config #25: running SBFspot failed - CRITICAL: bthConnect() returned -1

Matze79Berlin, 2020-11-10:
Sbfspot-config stops at the end of the installation process: 'running SBFspot failed'. The concole prints out 'CRITICAL: bthConnect() returned -1'. Looks like a bluetooth issue.

I'm using a RPi Zero W.
Inverter (SMA 3000TL) is being recognised in SBFspot (as well as in hcitool). 
Password not set (default 0000).
Inverter is in close distance.
BT connection attempts is set to maximum (15)

Tried V3.6 as well but the issue remained the same.

How can I solve this?

sbf-, 2020-11-10:
Make sure no other device is connected at the same time (e.g. SunnyExplorer on Win PC)
For the rest I'm out of ideas...

sbf-, 2020-11-10:
In this stage SBFspot is installed. You can run it manually with debug options:
`/usr/local/bin/sbfspot.3/SBFspot -finq -nosql -nocsv -d5 -v5`

Matze79Berlin, 2020-11-12:
> In this stage SBFspot is installed. You can run it manually with debug options:
> `/usr/local/bin/sbfspot.3/SBFspot -finq -nosql -nocsv -d5 -v5`

Thanks a lot for your swift reply. Yes, SBFspot is up and running. Unfortunately no bluetooth-connection is possible - there was an old pc that was connected to the inverter before but it has been replaced by the RPI Zero W. I suppose there's a communication issue involving the inverter.

Matze79Berlin, 2020-11-12:
Ok, problem was solved. The whole system setup was resetted and is working as described now incl. upload to pvoutput.org.

closed 2020-11-12 by Matze79Berlin