SBFspot/sbfspot-config #26: Failed to connect with inverter

predoli, 2020-11-11:
Hi! I use an external usb-bt adapter with a raspberry pi 3 but still I get the error that the connection failed.
I'm able to pair, scan and trust my inverter using bluetoothctl. Is my inverter not supported?
I can also find the inverter in the config scan tool.

/usr/local/bin/sbfspot.3/SBFspot -finq -nosql -nocsv -d5 -v5
SBFspot V3.7.0
Yet another tool to read power production of SMA solar inverters
(c) 2012-2020, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with MySQL support
Commandline Args: -finq -nosql -nocsv -d5 -v5
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Configuration settings:
BTAddress=00:80:25:08:7D:7C
Password=<undisclosed>
MIS_Enabled=0
Plantname=DachOst
OutputPath=/home/pi/smadata/%Y
OutputPathEvents=/home/pi/smadata/%Y/Events
Latitude=51.51
Longitude=-0.13
Timezone=Europe/London
CalculateMissingSpotValues=0
DateTimeFormat=%d/%m/%Y %H:%M:%S
DateFormat=%d/%m/%Y
TimeFormat=%H:%M:%S
SynchTime=7
SynchTimeLow=60
SynchTimeHigh=3600
SunRSOffset=900
DecimalPoint=comma
CSV_Delimiter=semicolon
Precision=3
CSV_Export=0
CSV_ExtendedHeader=1
CSV_Header=1
CSV_SaveZeroPower=1
CSV_Spot_TimeSource=0
CSV_Spot_WebboxHeader=0
Locale=en-US
BTConnectRetries=15
SQL_Database=SBFspot
SQL_Hostname=localhost
SQL_Username=pvmonitor
SQL_Password=<undisclosed>
### End of Config ###
Wed Nov 11 18:27:34 2020: INFO: Starting...
sunrise: 07:13
sunset : 16:15
Connecting to 00:80:25:08:7D:7C (1/15)
Initializing...
SUSyID: 125 - SessionID: 949573520 (0x38995790)
getPacket(2)
MAX_CommBuf is now 18 bytes
Received 18 bytes
Received 9 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1B 00 65 7C 7D 08 25 80 00 
00000010: 00 00 00 00 00 00 02 00 00 03 
00000020: 70 00 01 01 00 00 00 
cmd=2
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1B 00 65 7C 7D 08 25 80 00 
00000010: 00 00 00 00 00 00 02 00 00 03 
00000020: 70 00 01 01 00 00 00 
<<<=================================>>>
MAX_pcktBuf is now 27 bytes
SMA netID=01
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 00 00 00 00 00 00 
00000010: 7C 7D 08 25 80 00 02 00 00 04 
00000020: 70 00 01 00 00 00 00 01 00 00 
00000030: 00 
31 Bytes sent
getPacket(5)
Received 18 bytes
Received 13 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 7C 7D 08 25 80 00 
00000010: 00 00 00 00 00 00 0A 00 7C 7D 
00000020: 08 25 80 00 01 14 71 DA 7D 1A 
00000030: 00 
cmd=10
Received 18 bytes
Received 16 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 22 00 5C 7C 7D 08 25 80 00 
00000010: 00 00 00 00 00 00 05 00 7C 7D 
00000020: 08 25 80 00 01 01 14 71 DA 7D 
00000030: 1A 00 02 01 
cmd=5
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 22 00 5C 7C 7D 08 25 80 00 
00000010: 00 00 00 00 00 00 05 00 7C 7D 
00000020: 08 25 80 00 01 01 14 71 DA 7D 
00000030: 1A 00 02 01 
<<<=================================>>>
MAX_pcktBuf is now 34 bytes
Local BT address: 00:1A:7D:DA:71:14
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 3F 00 41 14 71 DA 7D 1A 00 
00000010: FF FF FF FF FF FF 01 00 7E FF 
00000020: 03 60 65 09 A0 FF FF FF FF FF 
00000030: FF 00 00 7D 5D 00 90 57 99 38 
00000040: 00 00 00 00 00 00 02 80 00 02 
00000050: 00 00 00 00 00 00 00 00 00 00 
00000060: E3 30 7E 
63 Bytes sent
getPacket(1)
Received 18 bytes
Received 5 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 17 00 69 7C 7D 08 25 80 00 
00000010: 00 00 00 00 00 00 01 00 7E FF 
00000020: 03 60 65 
cmd=1
PacketLength=23
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E FF 03 60 65 
<<<=================================>>>
Invalid chk 0xFF00 - Found 0x6003
Wed Nov 11 18:27:34 2020: CRITICAL: Failed to initialize communication with inverter.

sbf-, 2020-11-11:
> Is my inverter not supported?

Sure it is, but you have an unreliable BT connection. Try to move RPi closer to the inverter.

predoli, 2020-11-29:
Thank you for your response.

Moved the py now directly next to the inverter. Still no change. Can someone recommend a pi/bt hardware combination that already worked with their specific inverter?

I am using:

- Raspberry Pi 3 Model B Rev 1.2
- CSL - USB Nano Bluetooth-Adapter V4.0 - Bluetooth Dongle Class 4.0 Technologie
- SMA Sunny Boy SB 5000TL-20

sbf-, 2020-11-30:
This RPi has also built-in Bluetooth, no?
Are you sure the external BT is used (this should work if it is close enough to the inverter)
See also [SBFspot #402](https://github.com/SBFspot/SBFspot/issues/402)

closed 2021-01-13 by sbf-