SBFspot/sbfspot-config #27: Can not define MQTT user and password

EddyBin, 2020-11-20:
Try to install the MQTT component, but can not define user and password.
Manual adjustment of the  SBFspot.cfg with nano not possible as the file gives an error "File "SBFspot,cfg" is unwritable in nano

EddyBin, 2020-11-20:
try to add the following to the interface at MQTT_PublisherArgs

-u username -P password -h {host} -t {topic} -m "{{message}}"

but "-u username -P password" disappears if I click <ok>

sbf-, 2020-11-20:
> Manual adjustment of the SBFspot.cfg with nano not possible as the file gives an error "File "SBFspot,cfg" is unwritable in nano

SBFspot.cfg is read-only for user pi

```
sudo nano /usr/local/bin/SBFspot.cfg
```

commit 414f5886 (2021-03-18): V1.2.0
Fix #28 SBFspot and MariaDB/MySQL (DB Creation MariaDB)
Add: MariaDB port number (See SBFspot Issue #7)
Fix #27 Can't define MQTT user and password
Fix #31 '!' character in password SMA saved in config but not loaded
closed 2021-03-18 by sbf-