SBFspot/sbfspot-config #29: Error "bad lexical cast: source type value could not be interpreted as target" - Installation aborted

Gunhawk007, 2021-01-13:
Hi,
I have just tried to install the latest version 3.7.1 using the script. 

curl -s https://raw.githubusercontent.com/sbfspot/sbfspot-config/master/sbfspot-config | sudo bash -s 3.7.1

It reads in the config and let me do the settings but on install I got the error reported in the subject line. Note that this
actually refers to the SBFspot program to be broken - not the sbfspot-config tool. 
I then retried installing the old 3.7.0 which still works perfect. Curious what broke the install on 3.7.1.

sbf-, 2021-01-13:
What options do you use? SQLite/MariaDB/PVoutput/MQTT?
I tested with SQLite/PVouput and it works.

ghoxley, 2021-01-26:
Have noticed the same issue and the same fix.

I realised that the logging from the inverter stopped and when troubleshooting, got this message from SBFSpot....

pi@raspberrypi:/usr/local/bin/sbfspot.3 $ ./SBFspot -v
SBFspot V3.7.1
Yet another tool to read power production of SMA solar inverters
(c) 2012-2020, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -v
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
bad lexical cast: source type value could not be interpreted as target

Roll back to 3.7.0 restored operation. Logging now OK. I do not have update to PVoutput enabled. Test of the SQLite database passed OK under 3.7.1

sbf-, 2021-01-26:
Seems to be an issue with line endings in date_time_zonespec.csv (CRLF instead of LF)
I'll fix it ASAP

ghoxley, 2021-01-26:
Thanks very much - appreciated.

sbf-, 2021-01-26:
Replaced `date_time_zonespec.csv`
Should be fixed now

closed 2021-01-26 by sbf-