SBFspot/sbfspot-config #32: SBFSpot Upgrade with Multiple Inverters

badger-github, 2021-03-01:
Hi 
Thanks for such an awesome application, been running fine since 2019 but today I decided to upgrade from 3.5.2 to 3.7.1 using the sbfspot-config.  I backed up DB and then ran the command from wiki.  I selected my existing cfg file and did not pay much attention to the values as they looked ok and it was a very quick upgrade.  On checking the logs afterwards it appears that two of the three inverters were not transferred over in both the SBFspot.cfg and SBFspotUpload.cfg files. 

I did see you back those files up (awesome job as I forgot) and so I just copied the details back but just wondered if this is something I did wrong or is it expected to manually add them in again?

Thanks again for your efforts on this project.

Pete

sbf-, 2021-03-04:
Thanks.

What exactly was not migrated?

badger-github, 2021-03-04:
Hi, Thanks for getting back to me.
  
Basically in SBFspot.cfg original was IP_Address=192.168.1.12,192.168.1.10,192.168.1.11
 and after upgrade was IP_Address=192.168.1.12

In SBFspotUpload.cfg original was PVoutput_SID=XXXXXXXXXX:YYYYY,XXXXXXXXX:YYYYY,XXXXXXXXXX:YYYYY
and after was PVoutput_SID=XXXXXXXXXX:YYYYY

Maybe was something I did wrong.

sbf-, 2022-08-27:
Fixed in V1.3.0 (ec5f277)

closed 2022-08-27 by sbf-