SBFspot/sbfspot-config #39: curl install 3.8.3 error

halfagascan, 2021-11-15:
curl -s https://raw.githubusercontent.com/sbfspot/sbfspot-config/master/sbfspot-config | sudo bash -s 3.8.3
SBFspot Configuration Tool V1.2.0
Running on Raspberry Pi 3 Model B Rev 1.2
Checking whiptail... Installed.
architecture=armhf
Raspbian GNU/Linux 11 (bullseye)
username=pi
userhome=/home/pi
Minimum SBFspot version supported by this tool is 3.5.0
Request to install version 3.8.3
Version 3.8.3 is not available for installation.
Select one of (empty)
curl -s https://raw.githubusercontent.com/sbfspot/sbfspot-config/master/sbfspot-config | sudo bash -s 3.5.2
not available
Any method to install to bullseye?

halfagascan, 2021-11-15:
The script works fine on buster install

sbf-, 2021-11-15:
I´m working on bullseye image. Will be ready soon.

halfagascan, 2021-11-15:
ok, thanks

sbf-, 2021-11-17:
Bullseye images added to latest release

closed 2021-11-17 by sbf-