SBFspot/sbfspot-config #41: MariaDB password ending with "!" is not saved

nilaz, 2022-03-04:
I'm not sure the headline is the real problem but looks like it.
My MariaDB installation requires me to create accounts with passwords including special characters. In my case I added a "!" at the end to my generated password. The installer did its job but then sbfspot failed to report any energy. The log contained 
> [03:13:30.284] ERROR: Failed to open database: Can't connect to MySQL server on '192.168.44.34' (115)

When I restarted the installer and reused the configuration file the "!" at the end of the password was removed.

I changed the password and added a "/" in the middle of the same password (without the "!" at the end) and after that it worked well.

Is it a bug or feature? ;-)

nilaz, 2022-03-05:
I think I need to reconsider what I wrote above.
Today I get the following error in SBFspotUpload20220305.log:
> [03:10:30.272] ERROR: Failed to open database: Can't connect to MySQL server on '192.168.44.34' (115)
> [03:10:30.272] INFO: Stopping Daemon...
> [03:10:40.372] INFO: SBFspotUploadDaemon Version 3.0.1

Yesterday I got many rows like this:
> [17:25:30.935] INFO: Uploading datapoint: 20220304,17:25,58620864,0,,,40.60,224.61 => OK (200)

After a reinstall of SBFspot with same configuration file the upload log looks like this:
>[09:59:28.822] INFO: SBFspotUploadDaemon Version 3.0.1
>[09:59:28.841] INFO: Starting Daemon...
>[09:59:29.508] INFO: Uploading 30 datapoints, starting with 20220305,06:55,58620864,0,,,28.60,223.33 => OK (200)
>[10:00:30.994] INFO: Uploading 8 datapoints, starting with 20220305,09:25,58624690,3492,,,42.00,224.08 => OK (200)
>[10:05:30.460] INFO: Uploading datapoint: 20220305,10:05,58627234,4032,,,45.70,224.99 => OK (200)
>[10:10:30.935] INFO: Uploading datapoint: 20220305,10:10,58627575,4092,,,46.20,225.13 => OK (200)

Something goes wrong over night as it looks.

I'm installing 3.9.3 when I run the installer script.

sbf-, 2022-03-05:
Log is OK, except for the "Failed to open..."
Is the connection to the server stable?

nilaz, 2022-03-05:
My view is that it is stable even if I understand your valid question. 
What I can't understand is that after a reinstallation it works for the rest of the day and then dies over night and do not wake up until I reinstall SBFspot. This is the situation for three days in a row now after upgrade to latest SBFspot. When I did the upgrade I saw that I could use my MariaDB on my NAS to minimize writing data to the local SD card on my Raspberry PI running SBFspot.
I can mention that I started to use mqtt to send updates to my Home Assistant which works great by the way. I thought I should report the problem I anyway to help others.

The problem I see is now a small problem to me since I want the energy generation visible in my Home Assistant which the mqtt function do.

sbf-, 2022-03-05:
You don´t have to reinstall, just run
sudo systemctl start SBFspotUpload
Sure, it is not normal the daemon dies during the night

commit 8599eb82 (2024-06-16): V1.3.6 (Fixes #41 #75)
closed 2024-06-16 by sbf-