SBFspot/sbfspot-config #42: SBFspot-config hangs up during searching for "Searching for existing configuration(s)..."

Bastelritius, 2022-04-04:
Hi,

I wanted to make an update and get the following problem after starting the SBFspot-config tool.

`pi@iobroker2:~ $ curl -s https://raw.githubusercontent.com/sbfspot/sbfspot-config/master/sbfspot-config | sudo bash                        SBFspot Configuration Tool V1.2.1
Running on Raspberry Pi 3 Model B Rev 1.2
Checking whiptail... Installed.
architecture=armhf
Raspbian GNU/Linux 10 (buster)
username=pi
userhome=/home/pi
Minimum SBFspot version supported by this tool is 3.5.0
Request to install latest version (3.9.4)
Searching for existing configuration(s)...
`
Here the program hangs up and nothing happens anymore.

I think the problem is the follwing:
The command: `sudo find / -name SBFspot.cfg` shows the following result: `find: ‘/run/user/119/gvfs’: Permission denied`
I searched a little bit and found out that the problem is the virtual device which I´m using for my RaspiBackup.
The command `sudo find / -xdev -name SBFspot.cfg` shows then the right result: `/usr/local/bin/sbfspot.3/SBFspot.cfg`

How can I fix the problem?

sbf-, 2022-04-10:
> How can I fix the problem?

You can download the script, edit it and run the modified version locally

Bastelritius, 2022-05-05:
Could you give me more instructions please. Which script do you mean. How can I run it locally?

sbf-, 2022-05-09:
Download [sbfspot-config](https://github.com/SBFspot/sbfspot-config/blob/master/sbfspot-config) to e.g. "/home/pi"
Make the changes you mentioned
Run it: `sudo /home/pi/sbfspot-config`

Bastelritius, 2022-05-09:
ok, works. 

Thanks

closed 2022-05-09 by Bastelritius