SBFspot/sbfspot-config #66: Unable to get list of available SBFspot versions for 'bullseye'

wil76, 2023-10-09:
why I'm running into this error ?

pi@buderus:/usr/local/bin/sbfspot.3 $ curl -s https://raw.githubusercontent.com/sbfspot/sbfspot-config/master/sbfspot-config|sudo bash
SBFspot Configuration Tool V1.3.4
Running on Raspberry Pi 2 Model B Rev 1.1
Checking whiptail... Installed.
architecture=armhf
Raspbian GNU/Linux 11 (bullseye)
username=pi
userhome=/home/pi
Error: Unable to get list of available SBFspot versions for 'bullseye'

sbf-, 2023-10-09:
> why I'm running into this error ?

No idea... I tested the command and it works.
Maybe some glitch at github. Can you retry it?

wil76, 2023-10-09:
It has been work in my environment too. 

But now I permanently running into this error.

 

Is there a way to restart SBFspot without sbfspot-config ?

sbf-, 2023-10-10:
> Is there a way to restart SBFspot without sbfspot-config ?

With "restart" you want to say upgrade existing installation or just reinstall the same version?

wil76, 2023-10-10:
because of the above error I'm not able to reinstall or upgrade I just want to run the cronjob.
SBFspot is installed on my raspberry but the crontab is not working.
I can run it once, however I want to have it running via cron.

wil76, 2023-10-10:
maybe a hardware problem with my Pi:
crontab is available, but I can't edit it.

pi@mypi:/usr/local/bin/sbfspot.3 $ crontab -e
**/tmp/crontab.FdKCdY: Read-only file system
Creation of temporary crontab file failed - aborting**
pi@mypi:/usr/local/bin/sbfspot.3 $ crontab -l

## SBFspot
*/5 6-22 * * * /usr/local/bin/sbfspot.3/daydata
55 05 * * * /usr/local/bin/sbfspot.3/monthdata

``

sbf-, 2023-10-10:
Corrupt SD-card most probably