SBFspot/sbfspot-config #67: Unable to get list of available SBFspot versions for 'bookworm'

wil76, 2023-10-21:
now I have created a new environment with Debian 11 I'm running into this issue.
how can I bypass this.
I assume, that the current version will run on bookwork

sbf-, 2023-10-22:
Unfortunately the cross compilation toolchain is not yet available for "bookworm" so I can't build the images.
In the meantime you can follow the manual installation guide [here](https://github.com/SBFspot/SBFspot/wiki/Installation-Linux-SQLite#%EF%B8%8F-installation)

jogi19, 2023-10-31:
Is there also a guide for manual installation using mysql ? I can not find it.

sbf-, 2023-11-12:
V3.9.7 updated for bookworm

⚠️ New installation command:
`sudo bash -c "$(curl -s https://raw.githubusercontent.com/sbfspot/sbfspot-config/master/sbfspot-config)"`

closed 2023-11-12 by sbf-