SBFspot/sbfspot-config #69: scripts monthdata, daydata doesn't work

Rammy61, 2024-01-18:
I am using your script file sbfspot-config-master for communicating with my SMA converter.
But I get no daily or monthly updates from the converter.
The Bluetooth connection works.
The log file keeps track on what is going on.
After viewing your code I think there are 2 little erros in your script file 'sbfspot-config-master.1.3.5', monthdata and daydata.
You find the same code lines in 'sbfspot-config-master.1.0.15', monthdata and daydata.

'sbfspot-config-master.1.3.5', daydata, line 1348:
your code, 'echo "$dir_program/SBFspot $p_daydata$mqtt_arg -cfg$dir_config/$sbfspotcfg \$1 \$2 \$3 \$4 \$5 &>>\$log"'

if I change it to:
        echo "$dir_program/SBFspot $p_daydata$mqtt_arg -cfg"
        echo "$dir_config/$sbfspotcfg \$1 \$2 \$3 \$4 \$5 &>>\$log"
or
        echo -e "$dir_program/SBFspot $p_daydata$mqtt_arg -cfg\n$dir_config/$sbfspotcfg \$1 \$2 \$3 \$4 \$5 &>>\$log"

'sbfspot-config-master.1.3.5', monthdata, line 1359:
your code, 'echo "$dir_program/SBFspot $p_monthdata -cfg$dir_config/$sbfspotcfg \$1 \$2 \$3 \$4 \$5 &>>\$log"'

change it to:
        echo "$dir_program/SBFspot $p_monthdata -cfg"
        echo "$dir_config/$sbfspotcfg \$1 \$2 \$3 \$4 \$5 &>>\$log"
or
        echo -e "$dir_program/SBFspot $p_monthdata -cfg\n$dir_config/$sbfspotcfg \$1 \$2 \$3 \$4 \$5 &>>\$log"

Then the scripts files monthdata and daydata update works.

But I get an error when the line '/$sbfspotcfg \$1 \$2 \$3 \$4 \$5 &>>\$log' is executed.
It produces an error line in the /var/log/sbfspot.3/PV1666_20240118.log file called '/usr/local/bin/sbfspot.3/daydata: line 8: /usr/local/bin/sbfspot.3/SBFspot.cfg: Permission denied'
There seems to be something wrong with the file permission.
I tryed changing the file rights to 664, but that doesn't solve the error.
I also change the owner of the sbfspot.3 directory and all its file to root:root, but that also doesn't solve the error.
Maybe you could solve the problem.

Please let me now if you can agree with my conclusion and changing the errors.

Regards.

Aussems Jean

sbf-, 2024-01-19:
You can't split the commandline...

 `$dir_program/SBFspot $p_daydata$mqtt_arg -cfg$dir_config/$sbfspotcfg $1 $2 $3 $4 $5 &>>$log`
expands to
`/usr/local/bin/sbfspot.3/SBFspot -v -ad1 -am0 -ae1 -cfg/usr/local/bin/sbfspot.3/SBFspot.cfg $1 $2 $3 $4 $5 &>>$log`

`-cfg` is an argument of SBFspot executable and needs a path to the config file

```
pi@raspi32:~ $ /usr/local/bin/sbfspot.3/SBFspot -?
SBFspot V3.9.7
Yet another tool to read power production of SMA solar inverters
(c) 2012-2022, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with MySQL support
SBFspot [-options]
...
    -cfg:filename.ext   Set alternative config file
...
```

Rammy61, 2024-01-19:
I change the daydata script code into:

#!/bin/bash
#
{
echo -en "[$( date +%Y-%m-%d:%H.%M.%S )] Cron job started ..."
log=/var/log/sbfspot.3/PV1666_$(date '+%Y%m%d').log
/usr/local/bin/sbfspot.3/SBFspot -v -finq -ad1 -am0 -ae1 -cfg/usr/local/bin/sbfspot.3/SBFspot.cfg $1 $2 $3 $4 $5 &>>$log
[[ $? -eq 0 ]] || echo -en " Error in script: ${BASH_SOURCE[0]} at line 6"
} >>~/smadata/cron.log

When cron executes the daydata script I get an error in the ~/smadata/cron.log file

sbf-, 2024-01-19:
> Why do I need the -cfg part.

You don't need it if SBFspot.cfg is in the same directory as SBFspot

What is the output if you execute
`/usr/local/bin/sbfspot.3/SBFspot -v -finq -ad1 -am0 -ae1 -cfg/usr/local/bin/sbfspot.3/SBFspot.cfg`

Rammy61, 2024-01-19:
Now the update process works and I get no error message in the ~/smadata/cron.log file.
Note: I only use the csv part of the program.
Maybe you could change the code in a way that you don't need the -cfg part in the csv mode only.

Rammy61, 2024-01-27:
A code solution for the problem would be:

`
# launch script for daydata (every 5 minutes)
	local daydata="$dir_program/daydata"
	{
		echo "#!/bin/bash"
		echo "#"
		echo "log=$dir_logs/${Plantname}_\$(date '+%Y%m%d').log"
		if [ "$CSV_Export" -eq 1 ]; then
			echo "$dir_program/SBFspot $p_daydata"
		else
			echo "$dir_program/SBFspot $p_daydata$mqtt_arg -cfg$dir_config/$sbfspotcfg \$1 \$2 \$3 \$4 \$5 &>>\$log"
		fi
	} >"$daydata"
	chmod +x "$daydata"

# launch script for monthdata (once a day)
	local monthdata="$dir_program/monthdata"
	{
		echo "#!/bin/bash"
		echo "#"
		echo "log=$dir_logs/${Plantname}_\$(date '+%Y%m').log"
		if [ "$CSV_Export" -eq 1 ]; then
			echo "$dir_program/SBFspot $p_monthdata"
		else
			echo "$dir_program/SBFspot $p_monthdata -cfg$dir_config/$sbfspotcfg \$1 \$2 \$3 \$4 \$5 &>>\$log"
		fi
	} >"$monthdata"
	chmod +x "$monthdata"
`

Is this acceptable for you?

sbf-, 2024-01-27:
> Is this acceptable for you?

No, because your solution only works if the .cfg file is in the same directory as the SBFspot executable.
A user is free to have the config in a different directory and then you need the `-cfg` option

[image]

Rammy61, 2024-01-27:
O.k. I added the -cfg part, so the user is free to have a config file in a different directory.

`# launch script for daydata (every 5 minutes)
	local daydata="$dir_program/daydata"
	{
		echo "#!/bin/bash"
		echo "#"
		echo "log=$dir_logs/${Plantname}_\$(date '+%Y%m%d').log"
		if [ "$CSV_Export" -eq 1 ]; then
			echo "$dir_program/SBFspot $p_daydata -cfg$dir_config/$sbfspotcfg"
		else
			echo "$dir_program/SBFspot $p_daydata$mqtt_arg -cfg$dir_config/$sbfspotcfg \$1 \$2 \$3 \$4 \$5 &>>\$log"
		fi
	} >"$daydata"
	chmod +x "$daydata"

# launch script for monthdata (once a day)
	local monthdata="$dir_program/monthdata"
	{
		echo "#!/bin/bash"
		echo "#"
		echo "log=$dir_logs/${Plantname}_\$(date '+%Y%m').log"
		if [ "$CSV_Export" -eq 1 ]; then
			echo "$dir_program/SBFspot $p_monthdata -cfg$dir_config/$sbfspotcfg"
		else
			echo "$dir_program/SBFspot $p_monthdata -cfg$dir_config/$sbfspotcfg \$1 \$2 \$3 \$4 \$5 &>>\$log"
		fi
	} >"$monthdata"
	chmod +x "$monthdata"
`
Is this acceptable for you.

Rammy61, 2024-01-29:
I think I pinned down the problem.
Everyday a new logfile is created, so it seems.
Fact is that there is NO new logfile created and >> doesn't create the file.
It returns with the error: 'bash: /var/log/sbfspot.3/PV1666_20240129.log: Permission denied'
But when it would be created the owner is root and not USER.
So when the daydata batch file want to access this file by executing the daydata batch file it is denied because the batchfile is owend by root.
The command is terminated with the error 'bash: /var/log/sbfspot.3/PV1666_20240129.log: Permission denied'
When I remove the '&>>$log' part from the command '/usr/local/bin/sbfspot.3/SBFspot -v -finq -ad1 -am0 -ae1 -cfg/usr/local/bin/sbfspot.3/SBFspot.cfg $1 $2 $3 $4 $5 &>>$log'
Then I get no error and the update proces works.

To tackle the problem I come-up with the following code:

`daydata batch-file contents could look like:

#!/bin/bash
#
log=/var/log/sbfspot.3/PV1666_$(date '+%Y%m%d').log
[[ -f "$log" ]] || { sudo touch $log; sudo chown pi:pi $log; }
/usr/local/bin/sbfspot.3/SBFspot -v -finq -ad1 -am0 -ae1 -cfg/usr/local/bin/sbfspot.3/SBFspot.cfg $1 $2 $3 $4 $5 &>>$log

monthdata batch-file contents could look like:

#!/bin/bash
#
log=/var/log/sbfspot.3/PV1666_$(date '+%Y%m').log
[[ -f "$log" ]] || { sudo touch $log; sudo chown pi:pi $log; }
/usr/local/bin/sbfspot.3/SBFspot -v -sp0 -ad0 -am1 -ae1 -finq -cfg/usr/local/bin/sbfspot.3/SBFspot.cfg $1 $2 $3 $4 $5 &>>$log
`
The only problem I see at the moment is the 'pi:pi' construction.
During Set-up the USER is 'root' so you should find a way that you can convert 'pi:pi' -> '$( whoami ):$( whoami )' during executing of the batchfiles because you don't know the USER 'pi' at that moment.

p.s.: The Positional parameters $1, $2, $3, $4, $5 are always empty when the daydata and monthdata batchfiles are executed, so why did you add them?

sbf-, 2024-01-29:
Sorry, but I really don't get your problem.... this script has been run hundreds of times by several users without issues...

> Fact is that there is NO new logfile created and >> doesn't create the file.

My guess is the permission on /var/log/sbfspot.3 directory. Owner should be pi:pi and daydata/monthdata is running as pi via cron

> The Positional parameters $1, $2, $3, $4, $5 are always empty when the daydata and monthdata batchfiles are executed, so why did you add them?

This is to run daydata/monthdata with extra arguments

Rammy61, 2024-01-30:
I am just trying to show you where the problems accur that I encounter in the hope that you can correct them in your future version. Ofcourse I also wishes that I don't have to spend so much time on the subject to get it up and running.

My  /var/log/sbfspot.3 dir is owend by root and not by pi.
My hole /var dir is owned by root.
In line 915 you create the dir_log directory and in line 616 you change the owner from root to username.
In lines 2540-2547 you get the username.
But when the line 616 worked then the owner should be changed.
Maybe the username isn't get right.

I am running this software on a RaspberryPI computer with the latest image,  bookwurm.

Rammy61, 2024-01-30:
Yes, I can see the user pi in your picture.
Yes, it should be line 916, it was a mistyping.

[image]

I understand that the solution is the changing of the owner ship of the dir /var/log/sbfspot.3 to user pi.
But what is the cause that the owner isn't changed by the line 916 thats the question.