SBFspot/sbfspot-config #76: Installation on other distro

Depechie, 2024-06-25:
I'm trying to run the install command on a raspberry pi with dietpi os and getting follow exception:

```
Running on Raspberry Pi 3 Model B Plus Rev 1.3
Error: Not running Raspbian nor Raspberry Pi OS
```

Is there a way to do this on a non raspbian os?

sbf-, 2024-06-25:
You're running 32-bit I suppose?

I tested on 64-bit and the installer works but SBFspot can't connect via bluetooth...
```
dietpi@DietPi:~$ sudo bash -c "$(curl -s https://raw.githubusercontent.com/sbfspot/sbfspot-config/master/sbfspot-config)"
SBFspot Configuration Tool V1.3.7
Running on Raspberry Pi 3 Model B Rev 1.2
Checking whiptail... Installed.
architecture=arm64
Debian GNU/Linux 12 (bookworm)
username=dietpi
userhome=/home/dietpi
Minimum SBFspot version supported by this tool is 3.5.0
Request to install latest version (3.9.11)
Searching for existing configuration(s)...
Reading settings from /usr/local/bin/sbfspot.3/SBFspot.cfg
Saving defaults to /var/tmp/sbfspot.3/sbfspot.defaults... Done
Performing some checks before installation... Done
Downloading sbfspot-sqlite-arm64-linux-bookworm.tar.gz
Copying files...
SBFspot.db exists
Checking database...
Schema Version=1
Saving existing SBFspot.cfg... Done
Configuring SBFspot... Done
Running SBFspot...
SBFspot V3.9.11
Yet another tool to read power production of SMA solar inverters
(c) 2012-2024, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 64 bit with SQLite support
Commandline Args: -v -ad0 -am0 -ae0 -sp0 -nocsv -nosql -finq -cfg/usr/local/bin/sbfspot.3/SBFspot.cfg
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Tue Jun 25 22:43:44 2024: INFO: Starting...
sunrise: 05:31
sunset : 21:59
Connecting to 00:80:25:XX:XX:XX (1/10)
...
Connecting to 00:80:25:XX:XX:XX (10/10)
Tue Jun 25 22:43:53 2024: CRITICAL: bthConnect() returned -1
Tue Jun 25 22:43:53 2024: INFO: Done.
Installation aborted
```

Depechie, 2024-06-28:
Yes indeed, this is 32 bit

Depechie, 2024-06-28:
The specs of the Raspberry Pi 3 Model B Plus Rev 1.3 tell me it is a 64bit version...

Depechie, 2024-07-08:
Running https://github.com/nakla/sbfspot with docker compose on the same dietpi distro works!

So not sure what the main install checks from the OS, but somehow this does not work.

Of course the container version has another base OS...