SBFspot/sbfspot-config #77: BT Scan does not work with the installation script

obduser, 2024-12-26:
Hello, I have set up an RPi3B+ V1.3 with Bookworm. No Bluetooth device is found using the installation script. BT is activated and is also recognized "sudo bluetoothctl". With “scan on” devices in the area are also detected. What possible solutions are there?

obduser, 2024-12-26:
"hcitool scan" don't work...

sbf-, 2024-12-26:
Do you find your inverter with `hcitool scan` ?

obduser, 2024-12-26:
Hello, no see #2

obduser, 2024-12-26:
Hello again, sorry, I solved the problem. The internal BT module of the RPi3B+ doesn't seem to be that powerful. I placed the RPi directly near the WR, now hcitool scan also shows the WR. But maybe this will help someone with troubleshooting, thank you